LJMUAstroecology/flirpy #103: .fff files created with split_seqs have error in data field

ayazlali, 2024-08-19:
I was able to decode my .seq file and could see the images in the preview folder.  However, the .tiff files are all black and when I used Fff() to open one of the generated .fff files variable explorer shows an error in the data field: ```**ERROR**: a bytes-like object is required, not 'str'```. When I run split_seqs it ends with this error: ```ValueError: buffer is smaller than requested size``` on ```\flirpy\io\fff.py", line 194, in get_image
    self.image = np.frombuffer(
                 ^^^^^^^^^^^^^^```, but it does generate the preview jpgs files fine.
Here is the complete log when I run ```split_seqs```:
```
(flirenv) PS C:\Users\AYAZlALI\Documents\flirpy-main> py .\scripts\split_seqs -i C:\Users\AYAZlALI\Documents\Flir_files\300mV_200mHz_22ohm-0020.seq
INFO:__main__:Loading: C:\Users\AYAZlALI\Documents\Flir_files\300mV_200mHz_22ohm-0020.seq
INFO:flirpy.io.seq:Splitting 1 files
  0%|                                                                                                | 0/1 [00:00<?, ?it/s]INFO:flirpy.io.seq:Splitting C:\Users\AYAZlALI\Documents\Flir_files\300mV_200mHz_22ohm-0020.seq into C:\Users\AYAZlALI\Documents\flirpy-main\300mV_200mHz_22ohm-0020
100%|██████████████████████████████████████████████████████████████████████████████████▊| 594/595 [00:04<00:00, 136.12it/s]
  0%|                                                                                                | 0/1 [00:04<?, ?it/s]
Traceback (most recent call last):
  File "C:\Users\AYAZlALI\Documents\flirpy-main\scripts\split_seqs", line 112, in <module>
    folders = splitter.process(files)
              ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AYAZlALI\AppData\Local\anaconda3\envs\flirenv\Lib\site-packages\flirpy\io\seq.py", line 151, in process
    self._process_seq(seq, folder)
  File "C:\Users\AYAZlALI\AppData\Local\anaconda3\envs\flirenv\Lib\site-packages\flirpy\io\seq.py", line 279, in _process_seq
    image = frame.get_radiometric_image(meta=meta)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\AYAZlALI\AppData\Local\anaconda3\envs\flirenv\Lib\site-packages\flirpy\io\fff.py", line 180, in get_radiometric_image
    image = raw2temp(self.get_image(), self.meta)
                     ^^^^^^^^^^^^^^^^
  File "C:\Users\AYAZlALI\AppData\Local\anaconda3\envs\flirenv\Lib\site-packages\flirpy\io\fff.py", line 194, in get_image
    self.image = np.frombuffer(
                 ^^^^^^^^^^^^^^
ValueError: buffer is smaller than requested size
```

jveitchmichaelis, 2024-08-19:
Please try installing the latest version from github (0.4) and see if that fixes the issue. Thanks!

ayazlali, 2024-08-20:
Thank you, @jveitchmichaelis ! I installed the latest version in a new environment and still had the same issue. Edited the original post to include the issue template. I hope this helps with diagnosing the problem and thanks a lot for your support.

jveitchmichaelis, 2024-08-20:
Hmm, if you're willing to send me a copy of the SEQ file I'd be happy to take a look (you can email me a link if it's sensitive).

Maybe flirpy isn't correctly determining the frame size, are you able to check if it's correct? Do you get a black image that's the correct size(?) - or perhaps the splitter creates the image but doesn't fill it with data because it crashes.

Are you able to extract data using the `Seq` API - without using the splitter script? It's possible the script needs to be updated.

ayazlali, 2024-08-28:
I was able to extract the images from the .seq file with the FLIR science file sdk, so I think my file is OK. The file is not sensitive, it's just the video of a resistor, I will try to attach it to the original post. How would I use the .seq API? I was going to figure out what it does by modifying the script. Is there an easier way to learn how the API works?

jveitchmichaelis, 2024-09-02:
Thanks for this and sorry for the delay, I'll check out the file you've uploaded!

jveitchmichaelis, 2026-04-08:
[image]