LJMUAstroecology/flirpy #107: grab() fails on Lepton 3.5 with telemetry decode error (IndexError)

Elham6796, 2025-03-26:
Hi, I'm using Flirpy with a **FLIR Lepton 3.5 (radiometric)** connected via a **PureThermal board** on a Raspberry Pi 4b to implement a data logger.

When calling `grab()`, I consistently run into a crash caused by `decode_telemetry()`, even when trying to disable telemetry.

WARN:0@0.201] global cap_v4l.cpp:913 open VIDEOIO(V4L2:/dev/video1): can't open camera by index
[ WARN:0@0.201] global cap.cpp:478 open VIDEOIO(V4L2): backend is generally available but can't be used to capture by index Traceback (most recent call last): ... File "lepton.py", line 134, in decode_telemetry res = struct.unpack("<2cII16x4h6xIh2xh8xhI4xhhhhhh64xI172x", image[-2, :]) IndexError: index -2 is out of bounds for axis 0 with size 1

## What I tried

- Set `camera.telemetry = False`
- Used `Lepton.find_video_device()`
- Attempted to manually grab frames using `_capture_video_frame()` and reshape to (120, 160)
- Still, `grab()` always calls `decode_telemetry()` unconditionally

## Could you plz guide me on what I should do?

jveitchmichaelis, 2025-03-26:
Grab unconditionally decodes telemetry because it's non-trivial to change any hardware settings, so we go for the default output from the board. That may change in the future, but currently I have no time to support this.

If you're on a Pi, what does the image look like? This seems to be the same regression we've seen with recent versions of OpenCV where the returned image is 1D and not 2D. Please try lowering your version of OpenCV and see if that fixes it.

jveitchmichaelis, 2025-04-10:
This should be fixed for future releases of OpenCV https://github.com/opencv/opencv/pull/27193

Until then, please downgrade to 4.9 and check if that works.

closed 2025-04-10 by jveitchmichaelis