LJMUAstroecology/flirpy #108: Correct use of "raw2temp" and nan errors

raprakashvi, 2025-03-27:
Hi, 
I am very new to using FLIR camera, and have a FLIR BOSON (20640A095-6CAAX) camera which I am attempting to use on an ubuntu machine.  I am able to interface with the PC and get good quality raw image (apart from the current issue with Boson with numpy version number).

But when I attempt to use raw2temp function, I receive :
 ` [[nan nan nan]
  [nan nan nan]
  [nan nan nan]

dtype: float64`

My meta data I received for initial testing is :
`# Meta data for temperature conversion (used by the Boson camera)
META_DATA = {
    "Atmospheric Trans Alpha 1": 0.006569,
    "Atmospheric Trans Alpha 2": 0.01262,
    "Atmospheric Trans Beta 1": -0.002276,
    "Atmospheric Trans Beta 2": -0.00667,
    "Atmospheric Trans X": 1.9,
    "Planck R1": 14364.633,
    "Planck R2": 0.010603162,
    "Planck O": -5753.0,
    "Planck B": 1385.4,
    "Planck F": 1.0,
    "Emissivity": 1.0,
    "IR Window Transmission": 0.96,
    "IR Window Temperature": 20.0,
    "Object Distance": 1.0,
    "Atmospheric Temperature": 20.0,
    "Reflected Apparent Temperature": 20.0,
    "Relative Humidity": 50.0,
}`

I am unsure how to go about it, and how to get a correct meta data configuration. My target is to mount this camera on a moving object (very slowly) and monitor the movement of heat spot as an object is heated over time. I was reading that the camera needs time to reach a thermal equilibrium, and wanted to understand the best practices and how to move forward, as the camera will be rigidly fixed to the moving body and the target body will be stationary but heat spot will move. 

Thanks a lot for your time.

jveitchmichaelis, 2025-03-27:
Probably a good idea to trace the processing to see where you start to get NaNs introduced. Is it in the raw image? Or is there something in the metadata that's causing it?

raprakashvi, 2025-03-27:
Thanks a lot. Slightly off temperatures are fine. When I run the raw image steps, I am able to get good images :
dtype=uint8, shape=(512, 640, 3), min=79, max=85 

and the raw2temp step is where it breaks.

jveitchmichaelis, 2025-03-27:
OpenCV is already used by flirpy for camera capture. You can try the commands in the other issue thread to enable tlinear - it's on the todo list to add to the library.

raprakashvi, 2025-03-27:
[image]

This result is when I am using my opencvcamera class , not the Boson class:

`#!/usr/bin/env python3
"""
Thermal Imaging Capture Script

This script demonstrates how to capture and process thermal images from a FLIR Boson camera.
You can use either the standard OpenCV VideoCapture (by specifying a device path) or the FLIR
Boson camera via flirpy by using the --use_flirpy flag.

Usage Examples:
    # Use default video device (e.g., /dev/video0) and display preview only:
    $ python3 thermal_capture.py

    # Use FLIR Boson camera via flirpy (if installed) and record video to file:
    $ python3 thermal_capture.py --use_flirpy --record --output boson_recording.avi

    # Record for a fixed duration (in minutes), without displaying the preview:
    $ python3 thermal_capture.py --record --duration 1 --no_display

    Note: Downgrading from opencv-python 4.11 to 4.9.0.80 and downgrading numpy 1.26.4 seems to have done the trick! Else flirpy wont work
    Meta data does not seem to be correct and also does not seem to work with the current values. Either calibrate or not use.
"""

import cv2
import numpy as np
import time
import argparse

# Try to import flirpy modules
try:
    from flirpy.camera.boson import Boson
    from flirpy.util.raw import raw2temp
    FLIRPY_AVAILABLE = True
except ImportError:
    FLIRPY_AVAILABLE = False

# Meta data for temperature conversion (used by the Boson camera)
META_DATA = {
    "Atmospheric Trans Alpha 1": 0.006569,
    "Atmospheric Trans Alpha 2": 0.01262,
    "Atmospheric Trans Beta 1": -0.002276,
    "Atmospheric Trans Beta 2": -0.00667,
    "Atmospheric Trans X": 1.9,
    "Planck R1": 14364.633,
    "Planck R2": 0.010603162,
    "Planck O": -5753.0,
    "Planck B": 1385.4,
    "Planck F": 1.0,
    "Emissivity": 1.0,
    "IR Window Transmission": 0.96,
    "IR Window Temperature": 20.0,
    "Object Distance": 1.0,
    "Atmospheric Temperature": 20.0,
    "Reflected Apparent Temperature": 20.0,
    "Relative Humidity": 50.0,
}

class ThermalCameraInterface:
    """Abstract interface for a thermal camera."""
    def read_frame(self):
        """Read a raw frame from the camera."""
        raise NotImplementedError

    def get_display_frame(self, raw_frame):
        """
        Process the raw frame to produce a displayable image.
        For example: normalize, convert to 8-bit, apply a color map.
[... 165 lines cut ...]
            else:
                # Assume the raw frame is already 8-bit grayscale.
                video_frame = raw_frame
                if len(video_frame.shape) == 3 and video_frame.shape[2] == 3:
                    video_frame = cv2.cvtColor(video_frame, cv2.COLOR_BGR2GRAY)
            video_writer.write(video_frame)

        if not args.no_display:
            cv2.imshow("Thermal Image", display_frame)
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break

        if duration_seconds is not None and (time.time() - start_time) > duration_seconds:
            print("Recording duration reached.")
            break

    # Release resources.
    if video_writer:
        video_writer.release()
    cam.release()
    cv2.destroyAllWindows()

if __name__ == "__main__":
    main()
`

raprakashvi, 2025-03-27:
I have a followup question, after reading and exploring, I found out that my camera is not radiometric. My current understanding is that the camera always shows relative heating (0 - 255) mapping in non-radiometric cases. So if my scene changes, my entire calibration would also change.

Does this make sense:
If i make my test setup such that there is a a heat source or blackbody with a thermocouple and i get the realtime temperature reading. The body will be in all the frame and i use that pixel value to calibrate every frame? 

Also, I have a small workaround for raw2temp which when i implement directly here, my image quality is improved though the raw2temp value is very off (>10 C)

[improved_vis.txt](https://github.com/user-attachments/files/19493204/improved_vis.txt)

[image]

jveitchmichaelis, 2025-03-27:
Temperature should be linear with pixel count. The problem is normally that as the camera itself heats/cools, until it reaches equilibrium, is that the sensor readout will vary because it's basically a very sensitive thermometer. If you can, use two temperature sources to do a two-point calibration.

raprakashvi, 2025-03-27:
Thanks. Is there a rough time for equilibrium in your experience (time range)? Good point. I can try setting it up to do a 2 point calibration.

Do you have any thoughts on the code? I am reshaping my image input as I am still receiving the flattened array. This was the recommendation I got from the official support (and I am grateful for the guidance):

> At that point you can test against a blackbody (with a known steady state FPA temperature) to understand the relation between your camera's output and temperature. This will be some form of non-linear relationship. Both the SDK and the GUI (which uses the SDK in the backend) can gauge the FPA temperature with the command bosonlookupFPATempDegCx10() or bosonlookupFPATempDegKx10() or the TFPA output in the Status Panel section of the Diagnostic Tools tab in the GUI.

Do you have experience with these functions? Or 2 body method is better ?

jveitchmichaelis, 2025-03-27:
I'd suggest doing a lit review on this as there is a lot out there, and more rigorous than what I can hand-wave. If you leave the camera going for 20-30 minutes it might be OK? But it also depends on things like your readout rate, how well insulated the housing is (and obviously you don't want it _too_ hot), etc. It's very empirical which is why it's such a pain for non-radiometric cameras.

e.g. https://www.mdpi.com/1424-8220/17/10/2173 (but just look on Google Scholar).

You can log the FPA temperature for a bit and see what the curve looks like. Flirpy can do this: https://github.com/LJMUAstroecology/flirpy/blob/49179b31ed36e83821de18f3f56e0c1e84b01477/src/flirpy/camera/boson.py#L860

raprakashvi, 2025-03-27:
Thank you. I will give it a go for now. Thanks. Though the issue is not directly solved, I should be okay for now to test few things out. Appreciate your help.

closed 2025-03-27 by raprakashvi