LJMUAstroecology/flirpy #109: Using Two FLIR Lepton 3.5 Cameras Simultaneously with raspberry pi 4b

Elham6796, 2025-03-27:
Thanks in advance for your prompt response; I appreciate it.

I'm currently working on a project where I need to connect two FLIR Lepton 3.5 sensors using PureThermal Mini boards to measure surface temperature in two directions. My setup uses the following:

- Raspberry Pi 4b
- FLIR Lepton 3.5 
- Python 3.11.2
- OpenCV 4.5.5.64
- flirpy 0.4.0

**Everything works fine when using one thermal camera**. However, when I try to use both simultaneously, I encounter the following error:

raise ValueError("Unknown level: %r" % level)
ValueError: Unknown level: ...

It seems that the issue is related to an invalid or unrecognised log level being passed to logging. basicConfig().

Could you please help me resolve this issue? I would greatly appreciate any suggestions for properly initialising and handling multiple thermal cameras with flirpy.

Thanks again
Elham

jveitchmichaelis, 2025-03-27:
Odd. Where is this being thrown, and can you find out what `level` is?

jveitchmichaelis, 2025-03-27:
Yeah two separate objects shouldn't cause trouble. They're separate devices underneath.