LJMUAstroecology/flirpy #110: Flirpy Lens Selection Boson

raprakashvi, 2025-04-01:
Hi, 
I have a non-radiometric Boson camera that I m using with Flirpy (Ubuntu 24) and for debugging on Windows (using FLIR Boson 3.0) app. I was checking few calibration and was able to get clear image without the edge halo after calibration of lens 1. But lens 0 in the Boson app still had the Halo issue. In the app I am able to choose a lens and stick to it for better quality of image. 

Is there a way to define a lens and select that using Flirpy? It will be very helpful. 

Thanks,

jveitchmichaelis, 2025-04-01:
From what you've described, what the UI is doing is saving a flat field and then applying it? How is this different from the built-in FFC? 

Otherwise you would need to provide some more info as it's not clear to me exactly what you're describing.

What do you mean by calibration here?

closed 2025-04-10 by raprakashvi

raprakashvi, 2025-04-10:
Thanks. I was getting a warm (higher temperature) ring at the pheriphery of the image frame.